team4909/2019-Deep-Space #12: Figure out git flow for 2019

roshanr10, 2019-01-07:
Just something to think over, we've had some differing thought processes, I saw the wiki posts and I think we should figure out the best way going forward.

ashwinc12, 2019-01-07:
I know several teams that have a Testing Branch and a Competition Branch, so everything must go through the testing prior to being merged into competition. Just one idea.

TechplexEngineer, 2019-01-07:
@ashwinc12 I like that idea. Can Master = Competition? Then we have a testing / dev branch

roshanr10, 2019-01-07:
Also to consider: how do we PR? -- Who reviews, how many approvals, who merges?

TechplexEngineer, 2019-01-07:
I would like to get @derrell input. I think we should be lean on process,
only impose it when really needed.

One thing for sure:
- Master should always be deployable/stable (meaning tested on a robot)

roshanr10, 2019-01-08:
fair enough, I'd be onboard with the tagging, we just need to make sure we explain it to other software kids because currently they're comfortable with branches and have yet to try tags/releases

my only gripe with going with master as bleeding edge and fully code-reviewed but not fully tested is we may end up writing a decent chunk of code while the robot is in the bag between competitions and it's easier to convey pulling master as up to date than a particular version. 

it certainly makes sense for industry and OSS but our development cycle is also significantly shorter and we go thru a large number of versions making it more difficult in my opinion.

so my proposal is "1 PRs issued to a branch other than master, which is then fully tested, and then merged into master" as you had mentioned

derrell, 2019-01-08:
I have no problem with that. Good argument. Let's create a `testing` branch to which PRs are made by each developer, and once it's fully tested there, we then merge it into `master`.

HOWEVER: That "fully tested" part is really important in relation to something else you said there. 
> we may end up writing a decent chunk of code while the robot is in the bag between competitions

How is the code fully tested without a robot to test it on? It is really, really bad practice to make a bunch of changes to code where you can't fully test it, and then just assume it works and use it in competition...

roshanr10, 2019-01-08:
yeah, that sounds good to me

I should've mentioned more context to that... we only have six hours of time out of bag time the week before the competition to be shared between the various subteams so we have to optimize our testing.
so in this case, it'd essentially be PRing several fixes at once into `testing` to test all at once. 

In past, we've tried to test everything together and then check individual branches if issues arose. of course, we would only PR to `master` if everything passed testing.

derrell, 2019-01-08:
We soooooooo need to be building a second robot, to work with after bagging. Even professionals do not write bug-free software on the first try. There needs to be a way to confirm that it works and get it tested.

I know there are complexities to doing so, keeping the two identical, and maybe the two don't need to be quite perfectly identical as long as the non-identicalities are well-known and the values in code that depend on those differences can be configured during that measly six hours.

roshanr10, 2019-01-11:
are we all set to close this by end of meeting tomorrow? I can work with someone to get this written up and posted in the wiki

closed 2019-01-14 by roshanr10

TechplexEngineer, 2019-01-15:
Please provide a link to the wiki article here for reference