team4909/2019-Deep-Space #2: Add in basic classes

ashwinc12, 2019-01-07:
Add in the following:
-OI.java
-RobotMap.java
-Subsystem folder w/ example
-Command folder w/ example

roshanr10, 2019-01-07:
I'd refer to last years moreover WPIs example, their usage of robot map doesn't follow encapsulation very well

TechplexEngineer, 2019-01-07:
All Sensors and Motor Controllers should be instantiated in the subsystem that they belong to. Robotmap should be a list of `Static Final` information used to instantiate the sensor or controller.

For example when creating a Spark Motor Controller on PWM port 5 for an arm. The PWM port number should be declared in RobotMap.java and the Spark Motor Controller should be instantiated in the arm subsystem.

roshanr10, 2019-01-07:
@TechplexEngineer what do you think of how we did it last year? `new Intake(new Spark(3), new Spark(4))` in Robot.java. I'm all for the list of inputs/outputs but I personally found liked this, I'm open to either though

TechplexEngineer, 2019-01-07:
I feel strongly that the `3` and `4` in your example @roshanr10 should be in robot map as 
```
public static final int INTAKE_LEFT_MOTOR_PWM = 3;
public static final int INTAKE_RIGHT_MOTOR_PWM = 4;
```

roshanr10, 2019-01-07:
I'm assuming you would want PID, setpoints, etc. within robotmap as well?

TechplexEngineer, 2019-01-07:
I would make a counter argument to put PID setpoints at the top of the class where they are used as 

```
private final double ELEVATOR_KP = 0.025;
private final double ELEVATOR_KI  = 0.000;
private final double ELEVATOR_KD = 0.000;
```

closed 2019-01-07 by ashwinc12