team4909/2019-Deep-Space #6: Start thinking about what subsystems might be needed

TechplexEngineer, 2019-01-07:
There has been some talk about what the robot might look like, what functions it might have and how those functions will work. To start there will definitely be at least a drivetrain subsystem.

Review some guidance on Subsystems:
https://github.com/FRCteam4909/2019-Deep-Space/wiki/Subsystems

roshanr10, 2019-01-07:
I think this is more of a whole team thing, not just software. That said, we'll start to take a look into it

TechplexEngineer, 2019-01-22:
@roshanr10 Are the notes from the discussion posted somewhere? If no can you post them on a wiki page then link them here.

roshanr10, 2019-01-22:
It was just written out at the meeting— Ashwin and I have copies, I’ll type
it up and post tomorrow