team4909/2019-Deep-Space #16: Driving in straight line using white tape

ashwinc12, 2019-01-09:
Continuing the discussion from today's meeting... comment ideas below

ashwinc12, 2019-01-09:
I found something that could potentially work. Using 6 infrared sensors spread apart evenly at the front of the robot, if the average of the left three sensors is less than the right, than move to the right and vice-versa. Just one option. 

[https://maker.pro/arduino/projects/make-line-follower-robot]

TechplexEngineer, 2019-01-10:
We have qty (4) of these that should be able to detect the tape on the floor. 
https://www.firstchoicebyandymark.com/fc-42ef-d2mpak-f4

https://www.youtube.com/watch?v=uPFfevfpMxs

roshanr10, 2019-01-10:
Something to keep in mind about line followers is that a lot of them assume you’re on the line... we should check the tolerances on the sensor and how close the driver has to get it before the sensors can take over.

roshanr10, 2019-01-11:
> We could use easier forms of blob detection with pre-programmed functions.
@sMyn42 could you clarify what you meant by these “pre-programmed functions”? blob detection for the line using vision?

derrell, 2019-01-15:
Let's not close issues until they get pulled into master, so we don't forget that they are pending. Reopening.

roshanr10, 2019-02-01:
closing for now as per prev. comment, please reopen if deemed necessary

closed 2019-02-01 by roshanr10