team4909/2019-Deep-Space #25: Incorporate Lidar with Frankenstein

ashwinc12, 2019-01-16:


TechplexEngineer, 2019-01-18:
One difference between Arduino and RoboRio is the Arduino has pullup resistors on the I2C bus. Did you have pullup resistors when testing?

Here is some sample code for the lidar lite v1.... I think we have the v3. Not sure of the differences.

https://github.com/kauailabs/kauaibotsfirst2010/blob/master/Source/2015/GizmoBot/src/org/usfirst/frc2465/GizmoBot/subsystems/LIDAR.java

https://www.chiefdelphi.com/t/lidar/142375

https://www.chiefdelphi.com/t/lidar-lite-help-please/148130/13

roshanr10, 2019-01-18:
Yep, it's the V3, here's the documentation link, if that helps: http://static.garmin.com/pumac/LIDAR_Lite_v3_Operation_Manual_and_Technical_Specifications.pdf

ashwinc12, 2019-02-03:
@roshanr10 What were you thinking of using the PID distance for?

TechplexEngineer, 2019-02-03:
Drive until you are x inches from the rocket?

roshanr10, 2019-02-03:
Well, now I’m curious, what else were you planning to do with the LIDAR?

ashwinc12, 2019-02-03:
If `lidar.getDistance()` was less than 5 cm or so (can be determined later),  then exit the LineFollow command and call a different command (Ex: place hatch, place ball, etc). Not sure if a PID would be necessary in this situation, in my opinion.

roshanr10, 2019-02-03:
ah, yeah, makes sense. so we'd be navigating with the line, but using LIDAR as a finish condition. I was under the impression that you were planning to use the LIDAR to finish driving up as soon as you found the line.

my question though is how quick can we line follow in comparison to LIDAR & gyro? for a straight line distance, would the latter be quicker than using the line? just throwing another option in the air to consider

ashwinc12, 2019-02-03:
The working linefollow in testing is fairly quick and we'll be gradually increasing the speeds as well in the upcoming days.

roshanr10, 2019-02-03:
can you define fairly quick? also, is increasing the speed based on theory? I'd be interested to see how quickly it can go without drifting.

What's the tolerance between the distance between the inside sensors and the width of the line?

roshanr10, 2019-02-03:
@ashwinc12 just something running thru my mind, but could we use the LIDAR to automate the alignment for the climb? that way we can determine distance from the wall, and react accordingly

ashwinc12, 2019-02-03:
Yes, I thought that was part of #57

roshanr10, 2019-02-03:
yeah, I wasn't sure of implementation details- just curious if it seemed viable based on your test results in the shop

closed 2019-02-06 by ShashJar