team4909/2019-Deep-Space #32: Connecting Arduino with Lidar

ashwinc12, 2019-01-18:
Resource Links:
https://github.com/FRCteam4909/2018-PowerUp/blob/master/src/main/java/org/team4909/bionicframework/hardware/core/Arduino.java
https://github.com/FRCteam4909/2018-PowerUp/blob/master/arduino/SLAVE_I2C/SLAVE_I2C.ino
http://first.wpi.edu/FRC/roborio/release/docs/java/edu/wpi/first/wpilibj/I2C.html

ShashJar, 2019-01-18:
https://static.garmin.com/pumac/LIDAR_Lite_v3_Operation_Manual_and_Technical_Specifications.pdf

ShashJar, 2019-01-18:
https://github.com/garmin/LIDARLite_Arduino_Library

roshanr10, 2019-01-19:
What’s the difference between this and #25?

derrell, 2019-01-19:
This one pertains to using the Arduino library to read from the LIDAR and then send that data to the Rio. The other was intended to read LIDAR directly with I2C on the Rio, which turned out not to work with the initial work, so was (at least temporarily) set on the back burner.

ShashJar, 2019-01-20:
Notes to self:
     - Breadboard with LIDAR and Arduino connections pre-wired is on table above PTC in cage. Solder 
     and test w/ code sometime during week of 01/20.
     - For testing code, may be useful to reference links in this issue thus far as well as those in 
     [https://github.com/FRCteam4909/2019-Deep-Space/issues/29](url).
     - Main thing to test as of 01/19 is sending LIDAR data to Rio and also the getDistance() method. 
     Arduino analyzes LIDAR data fine (has already been tested).
     - Following the above testing, implement the getDistance() method into Robot.java so that it can be 
     used continuously to make corrections and adjustments throughout match. Also consider 
     mechanical design of the LIDAR (where to place the sensor on the robot, how to orient it, etc.).

ShashJar, 2019-01-23:
Note to self based off of meeting 01/22/19:
- Compilation errors when trying to build code before deploying to Frankie
- Fix errors at 01/23 meeting and then test if correct LIDAR values are being sent to Rio with Frankie
- I2C wires in Frankie's Rio with crimps attached, rest of wiring is with breadboard/LIDAR/Arduino setup...if time allows and testing is successful solder everything together at 01/23 meeting

roshanr10, 2019-02-02:
@ashwinc12 given that we have it working directly with the Rio, do we plan to look into this or are we good to close?

ashwinc12, 2019-02-03:
Can move this to #25 ... ended up using PWM rather than Arduino

closed 2019-02-03 by ashwinc12