team4909/2019-Deep-Space #38: Prototype code to use current from PDP to limit motor

TechplexEngineer, 2019-01-21:
Given an intake that uses a motor to manipulate a cargo, we will want code that measures the current drawn by the motor and provide a limit to ensure we don't pop the ball. 

- [ ]  When a "current spike" is detected, slow the motor and or stop the motor.
- [ ] Research code needed to prevent a stalled 775 motor from blowing up in smoke (I believe 195 does this)

roshanr10, 2019-01-21:
If we’re using SPXs, as previously discussed, they offer their own built in mechanism for current limiting, for both peak and continuous draw.

I’m not sure of the intricacies of this functionality but it may be simpler than running a continuous loop on the Rio checking for current draw.

Just something to consider as we look more into this.

roshanr10, 2019-01-21:
Disregard my previous comment— I misread the SRX vs SPX disclaimer in the combined manual.

TechplexEngineer, 2019-01-21:
Using SRX's for this should work shouldn't it. What disclaimer did you read?

roshanr10, 2019-01-21:
SRXs will, SPXs don’t have the ability, but that’s a $100 difference. It’s mentioned in the documentation in several places, I’ll add a link later.

roshanr10, 2019-01-21:
https://phoenix-documentation.readthedocs.io/en/latest/ch13_MC.html

it’s mentioned on here, how the method exists but always returns 0 on the SPX.

do we want to consider using SRXs for just this functionality?

roshanr10, 2019-02-01:
While we plan to have an IR sensor to detect when we are in possession of a ball, I'd like to investigate using current limiting as a secondary failsafe.

ashwinc12, 2019-02-09:
@roshanr10 What's the status of this?

roshanr10, 2019-02-09:
Ready for testing on comp bot, it’s been PRed

roshanr10, 2019-02-10:
this seemingly works, but we have to hone down the exact numbers Monday

roshanr10, 2019-02-12:
some psuedocode I got from a discussion on a FRC channel:
```current = pdp.getCurrentOnChannel()
signal = 1.0 // your pwm value
if (current > limit)
    signal *= limit/current```

theoretically this'll work more effectively than what we have now, will test tomorrow

Vultimate1, 2019-02-12:
@roshanr10 I have the current limiting code using the newer IntakeSubsystem and I can PR it, if you want.

roshanr10, 2019-02-12:
Nah, we’re all set, I pulled in the relevant changes

roshanr10, 2019-03-11:
closing based on IR sensor success

closed 2019-03-11 by roshanr10