team4909/2019-Deep-Space #44: Investigate using IR Sensor (Intake)

roshanr10, 2019-02-01:
I would like to use the IR sensor for detecting ball possession to avoid running the motors and possibly damaging game pieces. 

See: https://cdn-shop.adafruit.com/product-files/1031/GP2Y0A02YK-DATA-SHEET.PDF

So far, I've gotten a sensor to provide analog values back, we need to be able to translate that into usable values. The primary thought process is that we can use it similar to a garage door tripwire to detect when the voltage changes, we need to experiment with how effective that may be. 

The intake sides are spread apart by 16in.

roshanr10, 2019-02-01:
See referenced comments in #47

roshanr10, 2019-02-01:
Also side note, the real intake is 16" wide, while the prototype one is 17", minor details, but just something to remember to recalibrate.

ashwinc12, 2019-02-03:
Task was completed... now in testing phase in #43

closed 2019-02-03 by ashwinc12