team4909/2019-Deep-Space #60: Decide on Dashboard GUI

roshanr10, 2019-02-03:
We need to look more into the dashboard interface, currently planning to use the MS Surface we're getting for a touchscreen operator interface.

Derrell has written up a sample which binds directly to the Rio end in Java. However, we've also tested this in the past and found FRCDashboard a viable solution that connects directly to NT, which may be easier in terms of integration: https://github.com/FRCteam4909/2018-PowerUp/issues/11

derrell, 2019-02-03:
Indeed. If SmartDashboard can be sufficiently customized and become sufficiently easily usable, it's a fine tool to use. I'm currently working on the assumption that it will not be adequate, so developing an alternative. One of the big issues that I don't think SmartDashboard addresses is that we will have multiple modes of operation, some that are only very rarely used. In the UI that I'm thinking about, there are multiple tabs so it is easy to isolate entire subsections of user interface to things are are used together, which can avoid accidental incorrect choices.

Most of this can be done concurrently without slowing down progress. Certainly, go ahead and test each command using SmartDashboard. We'll also have the ability to run it with this alternate UI that I'm playing with. We can then decide which is easiest for the operator to use.

derrell, 2019-02-03:
One thing to consider is that the camera output will almost certainly be via SmartDashboard. If we have an external monitor for the camera view, can SmartDashboard be split into two parts, or can the camera output be separated and moved onto a different screen?

roshanr10, 2019-02-03:
To clarify, I was referring to [FRCDashboard](https://github.com/FRCDashboard/FRCDashboard), a fully customizable third party Electron app which connects to the robot via NetworkTables, similar to the existing SmartDashboard.

We could implement an interface in any web framework we’d like, it’d just be using a more standard communication stream behind the scenes so we don’t to recreate that.

I’m all for going with an HTML/JS dashboard if that’s what you’re looking for— also something to keep in mind is that a couple of our team members, myself included, have existing experience with bootstrap so we could put a tabbed dashboard in that rather quickly.

roshanr10, 2019-02-03:
This project also supports built-in camera stream access, and could be split however we’d like (open 2 tabs). I don’t believe you can split the official SmartDashboard as easily.

derrell, 2019-02-04:
I haven't used FRCDashboard before. It appears that it, too, uses an external framework (Electron), so also requires setting up its development environment... and also whatever JavaScript framework one wants to use (e.g., bootstrap). I like the fact that they're using Network Tables for the communication. I opted against that only because the documentation for it is so poor, but using Network Tables is more in line with standard practices than the HTTP server on the robot that I developed. 

Regardless, I'm coming up with _something_ that can be used. Time is very short. There is no requirement that anything I'm working on be used, but it will be available if there isn't time to do anything else.

roshanr10, 2019-02-04:
Ashwin and I have been able to setup FRCDashboard before, and we should be able to get it going again. Electron apps can be easily compiled to a native app, like Atom. There’s documentation on how to do this even within the FRCDashboard repo.

I really appreciate your efforts on this, I think I’ll work on this in parallel and we can decide when it comes to it.

roshanr10, 2019-02-04:
Threading onto this, I was able to confirm that we can call commands through NT in a manner consistent with my earlier response. I've taken a snapshot of the NT server data, and will play with it - hopefully I'll have something basic coordinated by tomorrows meeting

roshanr10, 2019-02-09:
@ashwinc12 how many cameras do we plan to be running, and how many simultaneously?

ashwinc12, 2019-02-10:
^ @roshanr10 2 cameras running simultaneously at 20 fps

roshanr10, 2019-02-10:
see my comments on cameras in #55

roshanr10, 2019-02-10:
Closing and merging into #55.

closed 2019-02-10 by roshanr10