team4909/2019-Deep-Space #63: Look into Gyro Mounting

roshanr10, 2019-02-04:
NavX Omnimount: https://pdocs.kauailabs.com/navx-mxp/installation/omnimount/

If we're planning to use the NavX at an atypical angle, we need to use this calibration procedure... that said, one of the axes has to be perpendicular to the surface, which it currently is not.

Just something to think about, if we intend to use the gyro. Thoughts/input?

ashwinc12, 2019-02-09:
Let's see if we can use this in conjunction with #56 in order to keep both the front and back stilts moving at the same speed

roshanr10, 2019-02-09:
I’d look at using encoders for this, I’m not sure if you saw my post in the
duplicate issue of this you created, but the navX will not work in the
position it’s currently mounted

derrell, 2019-02-09:
We are going to remount the Rio on its angled board so that it is horizontally oriented. The gyros will then work properly, and should provide better control than encoders. (There are two entirely different mechanisms for the front and rear stilts, so getting encoders to align them would be a complicated math problem. Certainly doable, but I think the gyro may be substantially easier to get working.

roshanr10, 2019-02-11:
Tested, working— closing

closed 2019-02-11 by roshanr10