team4909/2019-Deep-Space #66: Reflective tape detection with OpenCv

sMyn42, 2019-02-04:
Determine whether or not to use the Pi, and figure out what types are retuned by common OpenCV functions in Java.

The end goal is to be able to incorporate error from the angled tape camera into a PID.

sMyn42, 2019-02-04:
The camera stream issue is done, we have 2 cameras running on decent resolution, and 20 fps.

Vultimate1, 2019-02-08:
I have code for the GRIP reflective tape detection that returns a value indicating how much the robot should turn to line up with the tape, and follow the line.  I just need to test it tomorrow.

derrell, 2019-02-08:
Nice work!

Vultimate1, 2019-02-09:
Anybody know if it is possible to transfer the GRIP program output to the SmartDashBoard to display it?

roshanr10, 2019-02-09:
You'd have to use CameraServer to push the stream to the robot if I remember correctly, @sMyn42 may have a better idea

TechplexEngineer, 2019-02-09:
Where do you plan to execute your GRIP pipeline? On the roborio, on a
raspberry pi?

Once you answer that question we can discuss getting the data out of the
pipeline

sMyn42, 2019-02-10:
We decided to just stick to java and open cv, no Pi.

roshanr10, 2019-02-10:
@sMyn42 can you work with @Vultimate1 to display on the SmartDashboard during Monday's meeting?

Vultimate1, 2019-03-08:
Is vision processing still a possibility?  If so, we could just use the grip file created for detecting the reflective tape, without any java code.  We can still see the reflective tape through the grip file.  I've made it so you see its outlined shape to detect the tape.

roshanr10, 2019-03-08:
@Vultimate1 if you can get working *offboard*, then we can consider it. can you provide a more in depth explanation of your proposal?

roshanr10, 2019-03-11:
decided against.

closed 2019-03-11 by roshanr10