team4909/2019-Deep-Space #76: Needs to be tested

ashwinc12, 2019-02-09:
Running list of things that need to be tested on competition bot; feel free to add things to list

-Switch drive
-Line follow with an angled orientation
-Cargo intake commands (in, out, hold)
-Hatch panel intake commands (open, close)

derrell, 2019-02-09:
- Each low-level Command should be individually tested
- Each higher-level CommandGroup should be tested
- Manual control of each actuator (via, eventually, the secondary gamepad)

roshanr10, 2019-02-09:
Switch drive had been tested on comp bot, just an FYI

ashwinc12, 2019-02-09:
**Test Plan**

- [ ] Hatch Panel Open and Close (LB  & RB)

- [ ] Cargo Intake  (X & Y)

- [ ] Elevator Up and Down

- [ ] Elevator SetPoints

- [ ] Test NavX to see what values the `getPivot()` method prints in order to see if it is useable in current position

roshanr10, 2019-02-10:
@ashwinc12 what does "* Attach Photoelectric sensors in middle to LineFollow" mean?

roshanr10, 2019-02-10:
closing and breaking down into discrete issues.

closed 2019-02-10 by roshanr10