team4909/2019-Deep-Space #83: Design + Implement Driver User Interface

roshanr10, 2019-02-10:
Based on our current state of discussion, this is what we have done as a possible driver/operator interface for the robot this year. This issue entails discussing this with drive team, and then subsequently implementing the unfinished functionality.

# Gamepads
## Driver Gamepad:
- [ ] LY: Drive Speed
- [ ] RX: Drive Rotation
- [ ] LT: Drive Stilts
- [ ] LB: Retract Stilts
- [ ] RT: Invert Drive Direction
- [ ] RB: Line Follow
- [ ] X: Fast
- [ ] Y: Slow
- [ ] **_A: Floor Lines Camera???_**
- [ ] **_B: Look Ahead Camera???_**

## Operator Gamepad:
- [ ] LY: Elevator Arm Pivot
- [ ] RY: Elevator Speed
- [ ] RT: Cargo Intake In
- [ ] RB: Cargo Intake Out
- [ ] LT: Hatch Panel Intake In
- [ ] **_LB: Adjust Rocket Hatch Panel Setpoints for Cargo (Elevator & Arm Setpoint)???_**
- [ ] Y: Rocket Top Hatch Panel (Elevator & Arm Setpoint)
- [ ] X: Rocket Medium Hatch Panel (Elevator & Arm Setpoint)
- [ ] A: Cargo Ship/Human Player/Rocket Low - Hatch Panel (Elevator & Arm Setpoint)
- [ ] B: Cargo Ship - Cargo (Elevator & Arm Setpoint)

Notes:
- The elevator hatch panel setpoints are intended to set heights and wrist articulation.
- **_???_** signifies a potential but not guaranteed option

# Dashboard
## Main Screen (Jeff & Matt):
- [x] FRC Driver Station
- [ ] Match Timing
- [ ] Cargo Detection (Based on IR Sensors)
- [ ] Camera Streams
  - [ ] Facing Floor Lines/Ahead (Driver)
  - [ ] Facing Intake (Operator - Hatch & Cargo)
- [ ] Line Alignment (4x Photoelectric Sensors Feedback)
- [ ] Elevator Height (encoder ticks, for climbing)
- [ ] Drivetrain Inversion **_(graphic???)_**

## Operator Screen (Jeff):
- [ ] Match Timing
- [ ] Cargo Detection (Based on IR Sensors)
- [ ] Semiautonomous Commands Interface
  - [ ] AcquireHatchPanel
  - [ ] AcquireCargo
  - [ ] PlaceHatchPanel (Cargo Ship Low/Rocket Low)
  - [ ] PlaceHatchPanel (Rocket Med)
  - [ ] PlaceHatchPanel (Rocket High)
  - [ ] PlaceCargo (Cargo Ship Low)
  - [ ] PlaceCargo (Rocket Low)
  - [ ] PlaceCargo (Rocket Med)
  - [ ] PlaceCargo (Rocket High)
  - [ ] **_Climb ???_**
- [ ] Elevator Arm Angle (number & **_graphic_**)
- [ ] Intake Motor Current Reading

## Test Pane:
- [ ] LIDAR Distance
- [ ] Intake IR Sensor Values

# Major Design Concerns
## Driver Camera Stream(s)
I'm currently unsure if we want look ahead & floor lines cared for by the same camera or have a switching functionality for the stream... thoughts?

- [ ] Resolved Discussion

## Stilts Control Scheme
This is assuming my comments in #56 that the stilts automatically engage to keep the robot level once the elevator begins to lift the robot off the ground and that once `LB` is pressed, the PID loop is subsequently disabled.

However, this may or may not be the desired control scheme– this definitely needs more thought put into it.

- [ ] Resolved Discussion

## Control Scheme for Cargo Placement
I'm aware the manual control doesn't include discrete buttons for Rocket Cargo placement, but I think it's close enough at that point that we can easily manually maneuver it closer within the small gap, or use `LB` to correct to the cargo setpoint since they all maintain the same offset.

This is by design as adding other controls would add unneeded complexity given how this is intended to be a fallback and may necessitate more controllers.

- [ ] Resolved Discussion

derrell, 2019-02-10:
See my comment in #55 about software engineers designing the user interface. It's your (students') project. I highly recommend, however, that you take advantage of my experience, or better yet, the knowledge and experience of the user interface design expert we brought in, in conjunction with your actual users -- the driver and operator -- to come up with a good user interface design.

roshanr10, 2019-02-10:
Yes, I certainly agree, this can definitely use more work, and we’d love to hear input from you, Shelley, and the rest of the drive team. 

As I iterated in #55, this was just one potential possibility in which all of the outcomes are identified, and we can test with. Suggestions are welcome, and will of course be considered.

roshanr10, 2019-02-10:
Further, the dashboard section has already been reviewed and modified by Ashwin, Matt (driver) and Jeff (operator).

The gamepad section was designed based on earlier discussion and is pending further review — this is just a formal record of the result of all the discussions we’ve had to date, as we begin to approach finalizing the interface.

ashwinc12, 2019-02-11:
- Let's see if we can get more than just 2 cameras streaming as well even if we have to drop the fps down quite a bit... ideally, 3-4 would be nice now that we know we can stream 2
- If we were to stick with just two, I'd like to just have one that is at a medium height and can be used for solely driving and another camera facing the floor and has the red lasers within sight in order to be used for line following

roshanr10, 2019-02-11:
Even if we can get more than 2 running, should we..? I'm all for having for views but I'm not certain if we want them all streaming at the same time, for bandwidth concerns, and also the fact the drivers are only doing so much at a given point.

If there is one looking ahead and one facing the floor when aligning, does the driver ever need to be able to operate with both? Something we should get input from the drive team as well, but more isn't definitively better.

roshanr10, 2019-02-12:
We conferred with Jeff, Matt and Bryan at today's meeting, @ashwinc12 has their feedback and will post later, just an FYI

ashwinc12, 2019-02-13:
**Driver Comments**
- Would like to use Gyro with stilts
- Prefer the following on the touch screen:
  LB: Adjust Rocket Hatch Panel Setpoints for Cargo (Elevator & Arm Setpoint)???
  Y: Rocket Top Hatch Panel (Elevator & Arm Setpoint)
  X: Rocket Medium Hatch Panel (Elevator & Arm Setpoint)
  A: Cargo Ship/Human Player/Rocket Low - Hatch Panel (Elevator & Arm Setpoint)
  B: Cargo Ship - Cargo (Elevator & Arm Setpoint)
- Would like to use manual control for Climb

roshanr10, 2019-02-13:
- as for those on the touch screen, I'd be okay with it HOWEVER that means if the touchscreen does not work, they have NO way to use those setpoints. I'd want that in writing before we implement that... that also means if touch screen only semiautos don't work, they'd be using a gamepad for most of OI and then switching to the touch screen intermittently. if anything, both would be worth considering but I think we should touch base with them directly to verify that's what they want
- +1 on gyro w/ stilts
- what is "manual control"? I could give them direct access to the motor output if they wanted, but I don't think that's what they're looking for...

ashwinc12, 2019-02-15:
Should be noted that Jeff wanted LY for elevator and RY for the arm

ShashJar, 2019-02-15:
Will finalize controls on WIP tonight after double-checking w build

roshanr10, 2019-02-17:
@ShashJar have those controls been finalized in WIP?

ShashJar, 2019-02-17:
@roshanr10  haven't had time to test the robot with drive team and make sure they're good with all the controls, likely tomorrow since this doesn't take long

roshanr10, 2019-02-18:
@ShashJar @ashwinc12 what's the plan for dashboard currently? I'm fine with anything NetworkTables related, also I'd appreciate if you can do #94 whenever the code side of that is done – I can finish putting together custom dashboard easily if it comes to it... I have some local changes I'll push up whenever I get a chance

roshanr10, 2019-03-11:
all but #91 has been remedied, closing for now

closed 2019-03-11 by roshanr10