team4909/2019-Deep-Space #86: Test & Finalize LineFollowing Mechanism

roshanr10, 2019-02-10:


ShashJar, 2019-02-16:
Idea as a backup to Line Follow if testing does not yield efficient results:

- There will be a camera mounted on the front bottom of the robot facing downwards, or potentially angled slightly upwards like the photoelectric sensors currently are
- Drive team positions the robot near the line with some error being possibly up to 30 degrees on either side
- There will be two buttons on the robot. Each causes slow rotation of the robot in place to either the left or right
- The operator hits whichever button is necessary to begin the rotation, and they end the command with the same button when they see the gaffer tape lined up with the camera
- From there, it is just a matter of driving forward along the line

ShashJar, 2019-02-18:
To-Do for this Issue as of 02/17:

- Test current line follow command in conjunction with lidar (test separately as well)
- If current line follow command isn't very efficient, test out new command involving camera usage and defer to drive team for final decision
- Assign to controls necessary

roshanr10, 2019-03-11:
decided against.

closed 2019-03-11 by roshanr10