team4909/2019-Deep-Space #87: Working cameras on Robot

ashwinc12, 2019-02-13:
Test streaming two cameras on robot

sMyn42, 2019-02-13:
> to @sMyn42 <https://github.com/sMyn42>.

Ok boss

roshanr10, 2019-02-14:
has camera mounting been resolved?

ashwinc12, 2019-02-14:
Update: we were able to get 2 cameras streaming on competition bot yesterday. We are planning on having 4 cameras with the ability to switch between then/ have multiple up at once. As for mounting, we are 3D printing mounting cases.

roshanr10, 2019-02-14:
As for planning, has Flood been told.?

Also, what is each camera and each stream for, just trying think of how the
drivers might use them... another thing to consider is we need a USB hub if
we go over 2 cameras on the Rio. Doable, but just another thing to
consider, also unsure how much power it can provide and the cameras need.

derrell, 2019-02-14:
I suggest looking for, right away, a powered USB hub. The special requirement you have is that it be powered by 12V DC or 5V DC. Most powered USB hubs use a wallwart that may even provide 12V DC; you just need to confirm that in the specs before purchase. Let me know if you want confirmation or help.

ashwinc12, 2019-02-14:
Didn't get a chance to extensively talk with drive team so see where they wanted cameras and their opinion on how to mount. Later tonight I'll finalize where we are mounting each of our cameras and the mounting method for each. I'll mention the 3D printing to flood so he's in the loop

roshanr10, 2019-02-14:
An option: AmazonBasics 4 Port USB 3.0 Hub with 5V/2.5A power adapter
https://www.amazon.com/dp/B00DQFGH80/ref=cm_sw_r_cp_api_i_IOwzCb96BW15R

FYI Rios VRM 5V caps at 2A, so 12V means potentially higher amperage (20a)
or we need to use a buck converter of our own

I’d recommend ordering this by tonight so we can have it over the weekend,
if we can’t find one locally. Might be able to grab one from Best Buy or
Walmart if anyone can go.

roshanr10, 2019-02-14:
Tell him ASAP so he can print like now, you could have it for the meeting
then... print 6 or so

ashwinc12, 2019-02-14:
https://grabcad.com/library/microsoft-lifecam-hd-3000-mount-for-frc-vision-1

@TechplexEngineer and I looked at a different camera mount quickly last night because we couldn't find the old mount that we had before. However, we used this for the robot 2 years ago and it worked well. Thoughts?

roshanr10, 2019-02-14:
Yeah figure out the design but the last years mount is in the team onshape
org in a 3d printed parts folders, I just checked

I’ll take a look at this one shortly

ashwinc12, 2019-02-14:
Talked to Flood and he’s printing at least 4 out.

ShashJar, 2019-02-16:
02/15/19 Meeting: Locations for cameras have been chosen by drive team. 4 cameras in total. Code will be updated to lower resolution of each stream to fit each view and also place text specifying what angle each view is streaming from, so drive can practice with a convenient setup.

roshanr10, 2019-02-16:
Sounds good, are they having all 4 stream at once or switching
functionality?

roshanr10, 2019-02-16:
Also, can we get a clear list of what each camera angle is and it’s
intended use case in context of driving the robot?

ShashJar, 2019-02-18:
@sMyn42 please post progress and changes made

sMyn42, 2019-02-18:
The software side of this issue is done.

roshanr10, 2019-02-18:
Done and tested, or done in theory?

sMyn42, 2019-02-18:
Done.

roshanr10, 2019-02-25:
@sMyn42 has this been merged to master and ready for close?

closed 2019-02-25 by sMyn42

sMyn42, 2019-02-25:
Yes