team4909/2019-Deep-Space #88: Fix setpoint problem and adjust values for 7 different heights

ashwinc12, 2019-02-14:
Currently having trouble with the `setElevatorPosition` command:

The following method does not seem to be working (line 63 of elevatorSubsytem.java in WIP)
`public void setPosition(int position){
        leftSRX.set(ControlMode.Position, position);
    }
`
It was discussed that we may be missing a function that tells the elevator to run or that the elevator itself doesn't "know where it is".

Any troubleshooting comments would be helpful

roshanr10, 2019-02-14:
Does holding position still work.?

ashwinc12, 2019-02-14:
Yes, the arm and the elevator both stay up which means that they are staying at the setpoints.

ashwinc12, 2019-02-14:
https://www.chiefdelphi.com/t/talonsrx-position-control-mode-not-stopping/346268

Looks like a similar problem to ours

ashwinc12, 2019-02-14:
https://www.chiefdelphi.com/t/talonsrx-control-position-not-setting-elevator-position/346277

Got some responses back from the thread. Just committed the changes and will test at tomorrow's meeting.

ashwinc12, 2019-02-14:
Not sure how much it is going to change considering the setPosition() works with the operatorControl command

roshanr10, 2019-02-14:
I thought i changed the slots issues—I’d recommend using the default slot
0, just in case it affects anything else. I’m not sure why the code was
setup that way

roshanr10, 2019-02-14:
I'd highly recommend seeing the status output from the SRX using the Phoenix Tuner

roshanr10, 2019-02-14:
just for tracking history, part of #84

ashwinc12, 2019-02-14:
https://github.com/CrossTheRoadElec/Phoenix-Examples-Languages/blob/master/Java/PositionClosedLoop/src/main/java/frc/robot/Robot.java

roshanr10, 2019-02-14:
any luck?

ashwinc12, 2019-02-14:
We'll start with the test mode on the Pheonix Tuner and then go from there tonight. In the meanwhile, I'll look for other potential solutions.

roshanr10, 2019-02-14:
current working code is still in WIP branch, right?

ashwinc12, 2019-02-14:
Yes. It would be good to get that merged soon to testing

ashwinc12, 2019-02-15:
I think a potential solution would be to switch the encoder to absolute instead of relative. After doing some research, I found that the `.set()` method that we are using to set the elevator height either takes in as a parameter encoder ticks or an analog value, depending on the sensor. Prior, we were giving it a value of -2000 which doesn't make sense for an analog value. Changing the encoder to absolute should fix this problem.

roshanr10, 2019-02-15:
It uses whatever sensor is configured in the API... make sure you check the
requisite spec. let me know if you need help finding it

roshanr10, 2019-02-15:
I agree on absolute for arm, but we still need a way to zero the elevator
itself since there's no hard stops (unless they've since been added, if so
that's great)

ShashJar, 2019-02-16:
Problem has been fixed by making PID value nonzero. Still to-do with this issue:

- Test PID to make sure whatever value we choose is not random and will not negatively affect robot control
- Add all 7 needed setpoints for matchplay and test each - calibrate elevator speeds as necessary
- Figure out assignment of different setpoints to different buttons on controllers -----> speak to drive team about this, not previously decided in #83 
- Potentially implement a corresponding elevator arm command to make sure arm is perpendicular to elevator and ready to place panel/cargo following elevation

roshanr10, 2019-02-17:
@ShashJar arm is not always perpendicular to elevator– it depends on whether you're manipulating hatch vs. cargo

ShashJar, 2019-02-17:
@roshanr10 that's true - in general, we need to test whether we need commands for placing

ShashJar, 2019-02-18:
To-Do for this Issue as of 02/17:

- Ensure all setpoints are working well with the command to set position while the elevator is in any starting position
- Assign setpoints to corresponding button - consider an additional manipulator gamepad for elevator and climber if necessary

roshanr10, 2019-02-18:
as for extra gamepad, I'd recommend trying to avoid that b/c that may well increase cycle time

ShashJar, 2019-02-18:
public static final int elevatorSetpointHatchLow = 0; // Bottom hatch position
  public static final int elevatorSetpointHatchMiddle = -25300; // Middle hatch position
  public static final int elevatorSetpointHatchHigh = -51300; // Top hatch position

  public static final int elevatorSetpointCargoLow = -10582; // Bottom Cargo
  public static final int elevatorSetpointCargoMiddle = -30638; // Middle Cargo

We have not yet measured the high cargo and the cargo ship cargo (cargo ship hatch is the same level as low hatch on rocket)

roshanr10, 2019-02-25:
@ShashJar has this been merged to master and ready for close?

ashwinc12, 2019-02-25:
Have to re test these points.

roshanr10, 2019-03-14:
@ashwinc12 arm set points have been made relative, and changed– could use more testing tho

roshanr10, 2019-03-14:
do we need to work on carriage set points too?

roshanr10, 2019-03-14:
what does this issue actually entail at this point?

roshanr10, 2019-03-15:
@ashwinc12 @TechplexEngineer what else needs to be done for this to be considered closed?

roshanr10, 2019-03-18:
closing due to stale activity, will reopen subtasks so we can actually follow along with what's happening

closed 2019-03-18 by roshanr10