team4909/2019-Deep-Space #92: Map Gamepad Buttons for both controllers

ashwinc12, 2019-02-14:
- I know that the plan is to implement a user interface but I'd like to have the gamepads ready anyways

roshanr10, 2019-02-14:
The reasoning behind this is to cover edge cases if the touch screen fails-
we currently have no backup in that case.

Given what we have right now, and based on discussion yesterday, I think we
should be ABLE to fallback to any other team laptop and still be able to
compete successfully, while losing the semiautos... ideally, all else
should remain constant for the drivers.

Semi-autos are nice but robustness is also important, and gamepad are
rather easy to check off as a fallback on the way.

ashwinc12, 2019-02-18:
This seems to be done

closed 2019-02-18 by ashwinc12

roshanr10, 2019-02-18:
“Seems to be”?

roshanr10, 2019-03-13:
[image]
[image]
[image]