team4909/2019-Deep-Space #97: Set angle to 90 degrees for elevator arm

ashwinc12, 2019-02-16:


roshanr10, 2019-02-17:
again, what does this issue entail? the elevator, at least as far as I had heard, does not remain at 90º constantly while picking up or releasing cargo.

is this mapping UI to commands or getting the set point commands to work?

ashwinc12, 2019-02-17:
It was requested by Jeff to have a button that keeps the arm at 90 degrees in order to avoid the flailing and just make it generally easier to place cargo and hatches. Ideally using motion magic for the setpoints.

roshanr10, 2019-02-17:
MotionMagic is cool but it will only make it more difficult to tune. I'm still not convinced that PID is insufficient for our current needs.

Also, is Jeff aware that we have code that can keep the arm at whatever angle it's released at? That bit seems unclear and could drastically change his perspective– what is "generally easier"?

I would personally integrate arm setpoints with the elevator buttons as well, but I'd defer that to drive team. It makes sense to me to have the high hatch panel elevator button to also angle the arm for placing a hatch panel, wouldn't really be useful in nonstandard situations.

ShashJar, 2019-02-17:
Was a decision made on this front/has any progress been made on it?

roshanr10, 2019-02-18:
As far as I'm aware, none. See my addt. comments in #99. 

I'd also ask @TechplexEngineer and @derrell for their input, but I would strongly recommend against motion magic unless absolutely necessary. PID is pretty effective for basic mechanisms like the ones we have.

ShashJar, 2019-02-18:
To-Do for this Issue as of 02/17/19:
- Write command for placement of arm at 90 degrees to elevator
- Tune PID values to minimize flailing
- Assign command to button
https://robotics.stackexchange.com/questions/167/what-are-good-strategies-for-tuning-pid-loops

roshanr10, 2019-02-18:
additional PID reference: https://frc-pdr.readthedocs.io/en/latest/control/pid_control.html

ashwinc12, 2019-02-28:
Discussed upon and implemented

closed 2019-02-28 by ashwinc12