team4909/2019-Deep-Space #99: Implement motion magic

jasonbishop11, 2019-02-16:


roshanr10, 2019-02-17:
Precisely what does this issue entail? Motion Magic on what subsystem, and what advantage does it give us– is PID alone not acting quick enough? MM is a fabulous tool but I would check if it's truly necessary.

ashwinc12, 2019-02-17:
MM could be used for the elevator for setpoints because setting the position makes the elevator arm very wobbly. Motion magic will take into account the distance and the velocity to make this more streamlined with only a couple more values to tune. Check the motion-magic branch.

roshanr10, 2019-02-17:
if you have a proper PID, you should theoretically be fine.

MM will definitely help but tuning MM is even more difficult than PID... so my million dollar question for you: has it been written and tested thoroughly on the competition robot?

ShashJar, 2019-02-18:
PID will be tested further tomorrow for climber, elevator, and elevator arm. If we are unable to calibrate our subsystems to a certain point, we will attempt implementing MM. For now, #97, #88, and #56 are all concerned with PID tuning.

roshanr10, 2019-02-18:
Y'all are aware that MM is literally just MORE PID running at a higher interval right? You still need to tune PID to implement MM successfully.

MM is a 1D motion profile that essential develops PID set points for every 0.05s or something essentially you tell the mechanism where it should be when, and helps control for very precise movements by setting "checkpoint" along the way. What I'm trying to say is, if PID isn't tuned, MM won't help you.

ashwinc12, 2019-02-18:
Don't think it will be implemented due to working state of elevator setpoints... closing for now

closed 2019-02-18 by ashwinc12