team4909/2019-Deep-Space #133: Update WPILib Preferences.?

roshanr10, 2019-03-15:
https://github.com/FRCteam4909/2019-Deep-Space/blob/ffb3913942e681edf8d4e876ea4639253799f2d5/.wpilib/wpilib_preferences.json#L4

@TechplexEngineer thoughts on this? I don't think we've had any issues with this and this configuration is never used to the best of my knowledge. We have had this conversation before, but I'd just like to clarify with you.

TechplexEngineer, 2019-03-15:
I think it should be updated. Not sure if it is strictly needed but would be good to do

roshanr10, 2019-03-18:
@Vultimate1 should be pretty straightforward, copy the corresponding JSON file mentioned above from the starter project in place of the current one and create a pull request to testing

Vultimate1, 2019-03-19:
The JSON file in the starter project is the same as the current file.

roshanr10, 2019-03-19:
Hmmm that doesn’t seem right, did you update your local WPILib Plugins
installation?

roshanr10, 2019-03-19:
This is what I see in the latest update, can you fix and push?
```
{
    "enableCppIntellisense": false,
    "currentLanguage": "java",
    "projectYear": "2019",
    "teamNumber": 4909
}
```

Vultimate1, 2019-03-22:
I opened the wpilib_preferences.json file from GitHub and I saw this:
 {
  "enableCppIntellisense": false,
  "currentLanguage": "java",
  "projectYear": "Beta2019",
  "teamNumber": 4909
}
I also checked for updates, and there were none.  Also, this code is on the Testing branch.

Vultimate1, 2019-03-23:
Which branch were you looking at when you saw the code with "projectYear": "Beta2019"?

roshanr10, 2019-03-23:
pretty sure it's the same in all of them

TechplexEngineer, 2019-03-26:
This is NOT fixed.

https://github.com/FRCteam4909/2019-Deep-Space/blob/testing/.wpilib/wpilib_preferences.json#L4