team4909/2019-Deep-Space #158: Create Stilts PID Position Setpoint for Climb

roshanr10, 2019-03-18:
This would help us stay level even if we don't have gyro feedback. If this gets integrated into a command group that doesn't start climbing until we reach a threshold for "level", all the better

Done Includes
- [ ] Implementing Command to set stilts down so that the robot is roughly level
- [ ] Mapping aforementioned command to a button for the drivers, after discussing with them and Ashwin/Roshan
- [ ] Integrating Level Command into Command Group for Both Stilts/Elevator Down
- [ ] Regression Testing

Vultimate1, 2019-03-24:
How do we find the threshold level for the climber?  We don't seem to have any setpoints for it.

Vultimate1, 2019-03-24:
Are there any specific values for the level for the climber setpoint?